AUCOHL/Fault #2: Docker image issue

mdzakirhussain, 2019-11-22:
Hello,

**when I type in the terminal, the below shown** 

docker run -tiv `pwd`:`pwd` -w `pwd` --rm cloudv/fault:latest fault -V

 **I am getting the following error**

docker: Error response from daemon: OCI runtime create failed: container_linux.go:345: starting container process caused "process_linux.go:299: copying bootstrap data to pipe caused \"write init-p: broken pipe\"": unknown.

donn, 2019-11-22:
Hi: That's usually indicative of the fact you're probably on Linux, using an older kernel than is supported by the Docker image.

Would recommend trying Ubuntu 18.04.

donn, 2019-11-22:
Additionally, you can also update Docker to the latest possible version if you would like to keep your current operating system for the time being.

Please let me know how it goes.

mdzakirhussain, 2019-11-22:
> Hi: That's usually indicative of the fact you're probably on Linux, using an older kernel than is supported by the Docker image.
> 
> Would recommend trying Ubuntu 18.04.

Yes, I am using Ubuntu 18.04. Still facing the problem!!

donn, 2019-11-22:
Alright, I'm going to need way more details.

Please show me the outputs of `lsb_release -a` and `docker version`.

mdzakirhussain, 2019-11-22:
LSB Version:	core-9.20170808ubuntu1-noarch:security-9.20170808ubuntu1-noarch
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.3 LTS
Release:	18.04
Codename:	bionic

dock version
---
Client:
 Version:           18.09.9
 API version:       1.39
 Go version:        go1.13.4
 Git commit:        1752eb3
 Built:             Sat Nov 16 01:05:26 2019
 OS/Arch:           linux/amd64
 Experimental:      false

Server:
 Engine:
  Version:          18.09.9
  API version:      1.39 (minimum version 1.12)
  Go version:       go1.13.4
  Git commit:       9552f2b
  Built:            Sat Nov 16 01:07:48 2019
  OS/Arch:          linux/amd64
  Experimental:     false

donn, 2019-11-22:
@Manarabdelaty Mind running `docker version` too?

Manarabdelaty, 2019-11-22:
It prints the following: 

Client: Docker Engine - Community
 Version:           19.03.2
 API version:       1.40
 Go version:        go1.12.8
 Git commit:        6a30dfc
 Built:             Thu Aug 29 05:29:11 2019
 OS/Arch:           linux/amd64
 Experimental:      false

Server: Docker Engine - Community
 Engine:
  Version:          19.03.2
  API version:      1.40 (minimum version 1.12)
  Go version:       go1.12.8
  Git commit:       6a30dfc
  Built:            Thu Aug 29 05:27:45 2019
  OS/Arch:          linux/amd64
  Experimental:     false
 containerd:
  Version:          1.2.6
  GitCommit:        894b81a4b802e4eb2a91d1ce216b8817763c29fb
 runc:
  Version:          1.0.0-rc8
  GitCommit:        425e105d5a03fabd737a126ad93d62a9eeede87f
 docker-init:
  Version:          0.18.0
  GitCommit:        fec3683

mdzakirhussain, 2019-11-23:
After several tries,now this is happening 
when I type in the terminal> fault -V
I am getting this below shown

/usr/bin/fault: 7: [: -V: unexpected operator
Fault 0.2.0. ©Cloud V 2019. All rights reserved.

donn, 2019-11-24:
Thank you!

First, I recommend updating Docker to the latest version using this guide: https://docs.docker.com/install/linux/docker-ce/ubuntu/

The second issue is a bug I have introduced by accident in the latest version and I will push a fix posthaste. I apologize.

donn, 2019-11-24:
Alright, I think fixed the issue. Try now, please. 😅

mdzakirhussain, 2019-11-24:
I am clueless what really happening ??? I personally feel, if there is one script file which may look-into auto installation of all the required things will possibly be the best solution.

donn, 2019-11-24:
We're using Docker for that precise reason. Which, you can install from here: https://docs.docker.com/install/linux/docker-ce/ubuntu/ or by invoking this command: `curl -fsSL https://get.docker.com | sh`.

Docker runs a full environment with all dependencies preinstalled. It is like a virtual machine.

mdzakirhussain, 2019-11-24:
After typing in the terminal, the following

docker run -tiv `pwd`:`pwd` -w `pwd` --rm cloudv/fault:latest fault -V

it then starts downloading something and finally gives this following error

docker: failed to register layer: symlink ../c6331cd098b1793002eb5bf2b9f52b5807c55f763fb145ca9a621f4f5aab2541/diff /var/lib/docker/overlay2/l/FWJD4SB3EEXD453WLUC5VLI3FL: no such file or directory.
See 'docker run --help'.

mdzakirhussain, 2019-11-24:
If type the following in the terminal
fault -V
then I am getting this below error
/usr/bin/fault: 7: [: -V: unexpected operator
Fault 0.2.0. ©Cloud V 2019. All rights reserved.

mdzakirhussain, 2019-11-24:
my request is 
1. script file which installs all the following,( this script file user uses it and runs in the terminal )
       A. docker
       B. fault
       C. any other that is part of the application

end-user should able to only type the commands related to fault-application to use it.

mdzakirhussain, 2019-11-24:
My docker version
Docker version 19.03.5, build 633a0ea838
which is latest

donn, 2019-11-24:
Given the variance in operating systems, a script file that installs everything and ensures functionality is nigh on impossible, given that even containerized approaches such as Docker are prone to misconfiguration errors on part of the end user. We do not have the resources to commit to such a feature. Docker is the tool we elected to bypass this problem and we stand by it.

When a docker is image is updated, you have to manually update it by typing `docker pull cloudv/fault`.

Please, once again try these commands specifically:

```
docker pull cloudv/fault:latest
docker run -tiv `pwd`:`pwd` -w `pwd` --rm cloudv/fault:latest fault -V
```

I have tested and found these commands to work just now on all of:

* Apple macOS
* Amazon Linux (our build server)
* Lubuntu 18.04.

mdzakirhussain, 2019-11-25:
> Given the variance in operating systems, a script file that installs everything and ensures functionality is nigh on impossible, given that even containerized approaches such as Docker are prone to [...]

I am getting the following error
docker: Error response from daemon: invalid volume specification: '/:/': invalid mount config for type "bind": invalid specification: destination can't be '/'.
See 'docker run --help'.

mdzakirhussain, 2019-11-25:
> I am getting the following error
> docker: Error response from daemon: invalid volume specification: '/:/': invalid mount config for type "bind": invalid specification: destination can't be '/'.
> See [...]

upon typing the commands, I am getting the following

zakir@zakir-virtual-machine:~$ **sudo docker run -tiv `pwd`:`pwd` -w `pwd` --rm cloudv/fault:latest fault -V**
[sudo] password for zakir: 
Fault 0.2.1. ©Cloud V 2019. All rights reserved.
zakir@zakir-virtual-machine:~$ **fault -V**

Command 'fault' not found, did you mean:

  command 'faust' from deb faust

Try: sudo apt install <deb name>

mdzakirhussain, 2019-11-25:
This is another problem

zakir@zakir-virtual-machine:~$ EXEC_PREFIX=</usr/local/bin> sudo ./iverilog_build.swift
bash: ./iverilog_build.swift: No such file or directory

donn, 2019-11-25:
> upon typing the commands, I am getting the following
>
> zakir@zakir-virtual-machine:~$ sudo docker run -tiv `pwd`:`pwd` -w `pwd` --rm cloudv/fault:latest fault -V [sudo] password for zakir: Fault [...]

It seems there is a misconception. Docker is not a universal installer. Docker is a containerized environment, again, akin to a virtual machine.

You can run Fault commands by invoking:

```
sudo docker run -tiv `pwd`:`pwd` -w `pwd` --rm cloudv/fault:latest <command here>
```

Considering you got the intended output for `fault -V`, which is `Fault 0.2.1. ©Cloud V 2019. All rights reserved.`, means that this issue is particular issue is resolved.

> This is another problem
> 
> zakir@zakir-virtual-machine:~$ EXEC_PREFIX=</usr/local/bin> sudo ./iverilog_build.swift
> bash: ./iverilog_build.swift: No such file or directory

This has nothing to do with this particular issue. This is the IcarusVerilog build script, which you should not be running if you're using a containerized environment. The containerized environment prepares everything for you.

Nevertheless:

A. You are not in the correct folder. Your home folder has no file called `iverilog_build.swift`. You need to navigate to the repository first.
B. `</usr/local/bin>` is not a valid path. By UNIX convention, angled brackets typically indicate that this is a configurable path with a default value of `/usr/local/bin`. You can elect to leave it blank and just run `sudo ./iverilog_build.swift` assuming you are satisfied with `/usr/local/bin`.

closed 2019-11-25 by donn