AUCOHL/Fault #6: fault synth error

mdzakirhussain, 2019-11-25:
I was trying, but ending up with this following error. Please help

fault synth -t ha   --liberty /home/zakir/bin/FaultInstall/Tech/osu035/osu035_stdcells.lib ./ha.v 
Fatal error: 'try!' expression unexpectedly raised an error: Python exception: No module named 'pyverilog': file /share/Fault/.build/checkouts/PythonKit/PythonKit/Python.swift, line 683
Current stack trace:
0    libswiftCore.so                    0x00007fa999ecaec0 swift_reportError + 50
1    libswiftCore.so                    0x00007fa999f3bf60 _swift_stdlib_reportFatalErrorInFile + 115
2    libswiftCore.so                    0x00007fa999e5ff0e <unavailable> + 3514126
3    libswiftCore.so                    0x00007fa999e60087 <unavailable> + 3514503
4    libswiftCore.so                    0x00007fa999c5582d <unavailable> + 1374253
5    libswiftCore.so                    0x00007fa999e36e68 <unavailable> + 3346024
6    libswiftCore.so                    0x00007fa999c8d7a1 <unavailable> + 1603489
7    fault                              0x00005590de3f00df <unavailable> + 631007
8    fault                              0x00005590de3c17b2 <unavailable> + 440242
9    fault                              0x00005590de3afaf0 <unavailable> + 367344
10   libc.so.6                          0x00007fa997d50ab0 __libc_start_main + 231
11   fault                              0x00005590de3695ba <unavailable> + 79290
Illegal instruction (core dumped)

donn, 2019-11-26:
Sorry for the late reply. Just to confirm, is this a docker installation or did you install using `./install.swift`?

mdzakirhussain, 2019-11-27:
I have followed to install docker the way you had instructed. And it is successfully installed.
Thereafter I have to first download Fault and then from that Fault directory I run in the terminal ./install.swift

donn, 2019-11-27:
I should probably clarify this in the Readme: Docker is only useful if you don't want to install Fault on your computer.

Docker containers are, again, kind of like small virtual machines. They get created to run a single command, then they get destroyed when they're done. They're (usually) easier to set up, which is why I offer them as an option.

If you start the Docker container then try install Fault in there, that is redundant. The docker container `cloudv/fault` ***already has a copy of Fault preinstalled***.

That said, re-running ./install.swift should not do this. Let me give it a shot myself and try to diagnose why.

donn, 2019-12-04:
Sorry but could you try again now?

I replaced the Pyverilog installation with one that's a bit more stable in the docker container.

Remember to `docker pull cloudv/fault` to update. Thanks

donn, 2019-12-07:
No response, locked.

mdzakirhussain, 2019-12-07:
I am sorry for the delayed response. I am still busy with some assignments, possibly in another two days, I can test and will update.

donn, 2019-12-07:
That's okay!! Issue reopened. :)

mdzakirhussain, 2019-12-07:
Thank you, for reopening the issue.

mdzakirhussain, 2019-12-09:
Yes, It is now working. But, it is giving the following(Bold text), despite the following all commands are working perfectly.
--
zakir@zakir-virtual-machine:~$ fault synth --top ha /home/zakir/ha.v -o ha.synth  
Using Pyverilog v1.2.0
Generating LALR tables
**WARNING: 183 shift/reduce conflicts
mkdir: missing operand
Try 'mkdir --help' for more information.**

donn, 2019-12-09:
> WARNING: 183 shift/reduce conflicts
…is expected. It's a Pyverilog thing… the guys over there are apparently content with that warning. 😅

The broken mkdir is definitely not normal, though. Let me take a look at that.

donn, 2020-02-19:
Assuming stale at this point, try again and report any new errors. :)

closed 2020-02-19 by donn