AUCOHL/Fault #10: Fault simulation with external test pattern set ?

mdzakirhussain, 2020-03-16:
Could you please help how to do Fault simulation with the external test pattern set.

donn, 2020-03-16:
I don't think we support this feature yet? @shalan Do you have any thoughts?

Manarabdelaty, 2020-03-20:
You can now run fault simulations for an input test pattern set. Please check the wiki page for more details https://github.com/Cloud-V/Fault/wiki/Usage .

mdzakirhussain, 2020-03-20:
In the external test vectors json format file , how do we know the following
1. which set of vectors are covering SA0 and which set of vectors are covering SA1 faults 
2. And  which fault been covered on which port(s)

Manarabdelaty, 2020-03-20:
1. For each test vector, both the SA0 and the SA1 faults are simulated. So, you don't need to make this distinction in the input json file. 
2. After the simulations are concluded, Fault outputs the coverage information in a json file which specifies the covered SA0 and the SA1 faults for each test vector. 
For example, running this example in the wiki  [s27.v.tv.json](https://github.com/Cloud-V/Fault/blob/master/RTL/ISCAS_89/s27.v.tv.json) will output the coverage for the three test vectors in the following format : 
```json
"coverageList" : [
    {
      "vector" : [["+",0],["+",0],["+",0],["+",0],["+",0],["+",0],["+",1],["+",1],[ "+",0],["+",1],["+",0]],
      "goldenOutput" : "0010",
      "coverage" : {
        "sa1" : [
          "_14_.A",
          "_16_.B",
          "_15_.A",
          "GND",
          "_08_.A",
          "G1",
          "_19_.A",
          "_17_.Y",
          "_11_.Y",
          "_15_.B",
          "CK",
          "_09_.A"
        ],
        "sa0" : [
          "_17_.A",
          "_18_.A"
        ]
      }
    },
    {
      "vector" : [["+",0],["+",1],["+",0],["+",1],["+",1],["+",0],["+",1],["+",0],["+",1],["+",0],["+",0]],
      "goldenOutput" : "1100",
      "coverage" : {
        "sa1" : [
          "_16_.C",
          "_14_.B",
          "_13_.Y",
          "_17_.A",
          "_17_.Y",
          "_11_.Y",
          "_18_.A"
        ],
        "sa0" : [
          "_13_.B",
          "G1",
          "_19_.A",
          "_22_",
          "_15_.B",
          "_09_.A"
        ]
      }
    },
    {
"vector" : [["+",1],["+",0],["+",1],["+",0],["+",0],["+",1],["+",0],["+",0],["+",1],["+",1],["+",0]],
      "goldenOutput" : "1001",
      "coverage" : {
        "sa1" : [
          "_20_",
          "_13_.B",
          "_17_.A",
          "_15_.Y",
          "G0",
          "_22_",
          "_21_",
          "_21_.q",
          "_18_.A",
          "_15_.B",
          "_08_.Y",
          "_09_.A"
        ],
        "sa0" : [
          "_14_.A",
          "_16_.B",
          "_16_.C",
          "_14_.B",
          "GND",
          "_13_.Y",
          "G1",
          "_09_.Y",
          "_19_.A",
          "_17_.Y",
          "_11_.Y",
          "VDD"
        ]
      }
    }
  ]
```

mdzakirhussain, 2020-03-22:
Sorry for the delayed response. Ok, That is great, I will check that solution. Thanks, again.

closed 2020-03-22 by mdzakirhussain