AUCOHL/Fault #12: JTAG Commad causing an error

mahmoodulhassan-lm, 2020-12-18:
I tired to run the JTAG post insetion command but following error caused exiting the process. Is it bug or somwthing missing at my end?
```Generating LALR tables
WARNING: 183 shift/reduce conflicts
Creating top module\u2026
Stitching tap port...
Syntax error
Fatal error: 'try!' expression unexpectedly raised an error: Python exception: :8: before: library
Traceback:
  File "/usr/local/lib/python3.6/dist-packages/pyverilog/vparser/parser.py", line 2296, in parse
    ast = codeparser.parse()
  File "/usr/local/lib/python3.6/dist-packages/pyverilog/vparser/parser.py", line 2274, in parse
    ast = self.parser.parse(text, debug=debug)
  File "/usr/local/lib/python3.6/dist-packages/pyverilog/vparser/parser.py", line 77, in parse
    return self.parser.parse(text, lexer=self.lexer, debug=debug)
  File "/usr/local/lib/python3.6/dist-packages/pyverilog/vparser/ply/yacc.py", line 265, in parse
    return self.parseopt_notrack(input,lexer,debug,tracking,tokenfunc)
  File "/usr/local/lib/python3.6/dist-packages/pyverilog/vparser/ply/yacc.py", line 1047, in parseopt_notrack
    tok = self.errorfunc(errtoken)
  File "/usr/local/lib/python3.6/dist-packages/pyverilog/vparser/parser.py", line 2246, in p_error
    self._coord(p.lineno))
  File "/usr/local/lib/python3.6/dist-packages/pyverilog/vparser/plyparser.py", line 55, in _parse_error
    raise ParseError("%s: %s" % (coord, msg))
: file /share/Fault/.build/checkouts/PythonKit/PythonKit/Python.swift, line 602
Current stack trace:
0    libswiftCore.so                    0x00007f1d53268ea0 swift_reportError + 50
1    libswiftCore.so                    0x00007f1d532d9f40 _swift_stdlib_reportFatalErrorInFile + 115
2    libswiftCore.so                    0x00007f1d531fdeee <unavailable> + 3514094
3    libswiftCore.so                    0x00007f1d531fe067 <unavailable> + 3514471
4    libswiftCore.so                    0x00007f1d52ff37cd <unavailable> + 1374157
5    libswiftCore.so                    0x00007f1d531d4e48 <unavailable> + 3345992
6    libswiftCore.so                    0x00007f1d5302b741 <unavailable> + 1603393
7    fault                              0x00005644228eb09a <unavailable> + 1650842
8    fault                              0x000056442285e336 <unavailable> + 1073974
9    fault                              0x000056442286bcb3 <unavailable> + 1129651
10   fault                              0x000056442286bea1 <unavailable> + 1130145
11   fault                              0x0000564422882fd0 <unavailable> + 1224656
12   fault                              0x000056442286b943 <unavailable> + 1128771
13   libc.so.6                          0x00007f1d510edab0 __libc_start_main + 231
14   fault                              0x000056442277760a <unavailable> + 128522
Illegal instruction (core dumped)`````

mahmoodulhassan-lm, 2020-12-18:
> From what I can see it appears to be that Pyverilog (the syntax parser we're using) is detecting a syntax error for your code. If you feel comfortable doing so, please share the relevant Verilog netlist with us.
> 
> Regardless I don't believe Fault should crash if Pyverilog encounters a syntax error and I will try my best to fix that posthaste. Apologies for the inconvenience.

I am running the example given in the [wiki](https://github.com/Cloud-V/Fault/wiki/Usage) which is basically taking s27.v through the flow. I ran all the commands but the final [command](https://github.com/Cloud-V/Fault/wiki/Usage#jtag-interface-insertion) produced the aforementioned issue.

donn, 2020-12-19:
Alright, the issue has been fixed. Please try updating your Docker image by invoking `docker pull cloudv/fault`.

mahmoodulhassan-lm, 2020-12-19:
> Alright, the issue has been fixed. Please try updating your Docker image by invoking `docker pull cloudv/fault`.

Now the previous error has been resolved but a newer one has been produced.

`fault tap -l Tech/osu035/osu035_stdcells.lib -c Tech/osu035/osu035_stdcells.v --clock CK --reset reset Netlists/s27.netlist.v.chained.v
Generating LALR tables
WARNING: 183 shift/reduce conflicts
Creating top module…
Stitching tap port…
Generating LALR tables
WARNING: 183 shift/reduce conflicts
Generating LALR tables
WARNING: 183 shift/reduce conflicts
Resynthesizing with yosys…
Verifying tap port integrity…
Generating LALR tables
WARNING: 183 shift/reduce conflicts
An iverilog error has occurred: 
./Netlists/s27.netlist.v.chained.v.jtag.v:1565: error: Unknown module type: $_DFF_PP0_
./Netlists/s27.netlist.v.chained.v.jtag.v:1571: error: Unknown module type: $_DFF_PP0_
./Netlists/s27.netlist.v.chained.v.jtag.v:1577: error: Unknown module type: $_DFF_PP0_
./Netlists/s27.netlist.v.chained.v.jtag.v:1583: error: Unknown module type: $_DFF_PP0_
./Netlists/s27.netlist.v.chained.v.jtag.v:1589: error: Unknown module type: $_DFF_PP0_
./Netlists/s27.netlist.v.chained.v.jtag.v:1595: error: Unknown module type: $_DFF_PP0_
./Netlists/s27.netlist.v.chained.v.jtag.v:1687: error: Unknown module type: $_DFF_PP0_
8 error(s) during elaboration.
*** These modules were missing:
        $_DFF_PP0_ referenced 7 times.
`

donn, 2020-12-19:
@Manarabdelaty We really need to get rid of the automatic osu035 thing. It's causing far more headaches than it's worth at this point.

--

Instructions have been updated. A workaround for now is to explicitly provide fault synth and fault chain with the liberty file and the cell model file explicitly. The instructions have been updated in the wiki. I have went ahead and manually built the Docker container and verified these steps. Please verify that this works.

mahmoodulhassan-lm, 2020-12-20:
> @Manarabdelaty We really need to get rid of the automatic osu035 thing. It's causing far more headaches than it's worth at this point.
> 
> --
> 
> A workaround for now is to explicitly provide fault synth and fault chain with the liberty file and the cell model file. The instructions have been updated in the wiki. I have went ahead and manually built the Docker container and verified these steps. Please confirm that this works now and apologies for any trouble that this may have caused you.

The spcific issue of JTAG port isertion has been resolved. I really appreciate your cooperation. 
Thanks @donn

closed 2020-12-20 by mahmoodulhassan-lm