AUCOHL/Fault #13: Error while Running spm design with Sky130A library

mahmoodulhassan-lm, 2021-01-01:
I am trying to run [spm](https://github.com/Cloud-V/Fault/blob/master/Tests/RTL/spm.v) with Fault. Synthesis and Cut commands are successful but when I try to run 
```fault -c Tech/sky130A/libs.ref/sky130_fd_sc_hd/verilog/sky130_fd_sc_hd.v -v 100 -r 50 -m 95 --ceiling 1000 Netlists/spm.netlist.v.cut.v```
Which actually uses sky130_fd_sc_hd.v file, the following error appears. 
```Generating LALR tables
WARNING: 183 shift/reduce conflicts
Processing module CSADD…
Found 25 fault sites in 6 gates and 5 ports.
Performing simulations…
Skipped 84 duplicate generated test vectors.
./Tech/sky130A/libs.ref/sky130_fd_sc_hd/verilog/sky130_fd_sc_hd.v:40410: syntax error./Tech/sky130A/libs.ref/sky130_fd_sc_hd/verilog/sky130_fd_sc_hd.v:40410: syntax error
./Tech/sky130A/libs.ref/sky130_fd_sc_hd/verilog/sky130_fd_sc_hd.v:40410: error: invalid module item.

./Tech/sky130A/libs.ref/sky130_fd_sc_hd/verilog/sky130_fd_sc_hd.v:40410: error: invalid module item.
./Tech/sky130A/libs.ref/sky130_fd_sc_hd/verilog/sky130_fd_sc_hd.v:40410: syntax error
./Tech/sky130A/libs.ref/sky130_fd_sc_hd/verilog/sky130_fd_sc_hd.v:40410: error: invalid module item.
./Tech/sky130A/libs.ref/sky130_fd_sc_hd/verilog/sky130_fd_sc_hd.v:40410: syntax error
./Tech/sky130A/libs.ref/sky130_fd_sc_hd/verilog/sky130_fd_sc_hd.v:40410: error: invalid module item.```

The SKY130A is a standard library, so I am unable to find the issue. 
P.s: I have taken this library from [openlane](https://github.com/efabless/openlane).

donn, 2021-01-02:
Alright, in the future we'd intensely appreciate uploading the offending file(s) because the file in question is actually generated by [open_pdks](https://github.com/rtimothyedwards/open_pdks). It is not a "standard" file.

The issue will be tracked in https://github.com/RTimothyEdwards/open_pdks/issues/87. This may be an issue with any of IcarusVerilog, open_pdks or the Skywater PDK, but ultimately not Fault.

mahmoodulhassan-lm, 2021-01-02:
> Alright, in the future we'd intensely appreciate uploading the offending file(s) because the file in question is actually generated by [open_pdks](https://github.com/rtimothyedwards/open_pdks). It is not a "standard" file.
> 
> The issue will be tracked in [RTimothyEdwards/open_pdks#87](https://github.com/RTimothyEdwards/open_pdks/issues/87). This may be an issue with any of IcarusVerilog, open_pdks or the Skywater PDK, but ultimately not Fault.

@donn 
Thanks for the clarification.

mahmoodulhassan-lm, 2021-01-04:
@donn 
I commented out ' wire 1' on line 40410 in the file sky130_fd_sc_hd.v and included primitives.v file. Now a new error message is being shown.

 ```
fault -c Tech/sky130A/libs.ref/sky130_fd_sc_hd/verilog/sky130_fd_sc_hd.v -v 100 -r 50 -m 95 --ceiling 100 Netlists/spm.netlist.v.cut.v
Generating LALR tables
WARNING: 183 shift/reduce conflicts
Processing module CSADD…
Found 25 fault sites in 6 gates and 5 ports.
Performing simulations…
Skipped 84 duplicate generated test vectors.
[Warning]: golden output contains x or z.
[Warning]: golden output contains x or z.
[Warning]: golden output contains x or z.
[Warning]: golden output contains x or z.
[Warning]: golden output contains x or z.
[Warning]: golden output contains x or z.
[Warning]: golden output contains x or z.
[Warning]: golden output contains x or z.
[Warning]: golden output contains x or z.
[Warning]: golden output contains x or z.
[Warning]: golden output contains x or z.
[Warning]: golden output contains x or z.
[Warning]: golden output contains x or z.
[Warning]: golden output contains x or z.
[Warning]: golden output contains x or z.
[Warning]: golden output contains x or z.
Hit ceiling. Settling for current coverage.
Time elapsed: 3.61s.
Simulations concluded: Coverage 0.0%
```
I have attached sky130_fd_sc_hd.txt and primitives.txt for reference

[sky130_fd_sc_hd.txt](https://github.com/Cloud-V/Fault/files/5763246/sky130_fd_sc_hd.txt)
[primitives.txt](https://github.com/Cloud-V/Fault/files/5763250/primitives.txt)

donn, 2021-01-04:
Thanks for the report, we'll investigate

Manarabdelaty, 2021-01-04:
@mahmoodulhassan-lm 
The following macros have to be defined to be able to run the simulations with sky130
``
`define FUNCTIONAL 
``
``
`define UNIT_DELAY #1
``
Can you verify that these macros were set ? You can add them to sky130_fd_sc_hd.v

mahmoodulhassan-lm, 2021-01-04:
@Manarabdelaty when I entered both of the above definitions, the following error popped up.
```
fault -c Tech/sky130A/libs.ref/sky130_fd_sc_hd/verilog/sky130_fd_sc_hd.v -v 100 -r 50 -m 95 --ceiling 100 Netlists/spm.netlist.v.cut.v
Generating LALR tables
WARNING: 183 shift/reduce conflicts
Processing module CSADD…
Found 25 fault sites in 6 gates and 5 ports.
Performing simulations…
Skipped 84 duplicate generated test vectors.
./Tech/sky130A/libs.ref/sky130_fd_sc_hd/verilog/sky130_fd_sc_hd.v:19754: syntax error
I give up.
./Tech/sky130A/libs.ref/sky130_fd_sc_hd/verilog/sky130_fd_sc_hd.v:19754: syntax error
I give up.
./Tech/sky130A/libs.ref/sky130_fd_sc_hd/verilog/sky130_fd_sc_hd.v:19754: syntax error
I give up.
./Tech/sky130A/libs.ref/sky130_fd_sc_hd/verilog/sky130_fd_sc_hd.v:19754: syntax error
I give up.
```

Manarabdelaty, 2021-01-04:
I think it is because the `lpflow_bleeder` cells in the verilog views have syntax errors in the current version. I commented them out in the file you posted and was able to successfully pass it to Iverilog without syntax errors. 

[sky130_fd_sc_hd.txt](https://github.com/Cloud-V/Fault/files/5764642/sky130_fd_sc_hd.txt)

mahmoodulhassan-lm, 2021-01-04:
Please look into the output now. 
```
fault -c Tech/sky130A/libs.ref/sky130_fd_sc_hd/verilog/sky130_fd_sc_hd.v -v 100 -r 50 -m 95 --ceiling 1000 Netlists/spm.netlist.v.cut.v
Generating LALR tables
WARNING: 183 shift/reduce conflicts
Processing module CSADD…
Found 25 fault sites in 6 gates and 5 ports.
Performing simulations…
Skipped 84 duplicate generated test vectors.
[Warning]: golden output contains x or z.
[Warning]: golden output contains x or z.
[Warning]: golden output contains x or z.
[Warning]: golden output contains x or z.
[Warning]: golden output contains x or z.
[Warning]: golden output contains x or z.
[Warning]: golden output contains x or z.
[Warning]: golden output contains x or z.
Minimum coverage not met (4.0%/95.0%,) incrementing to 150…
Skipped 50 duplicate generated test vectors.
Minimum coverage not met (4.0%/95.0%,) incrementing to 200…
Skipped 50 duplicate generated test vectors.
Minimum coverage not met (4.0%/95.0%,) incrementing to 250…
Skipped 50 duplicate generated test vectors.
Minimum coverage not met (4.0%/95.0%,) incrementing to 300…
Skipped 50 duplicate generated test vectors.
Minimum coverage not met (4.0%/95.0%,) incrementing to 350…
Skipped 50 duplicate generated test vectors.
Minimum coverage not met (4.0%/95.0%,) incrementing to 400…
Skipped 50 duplicate generated test vectors.
Minimum coverage not met (4.0%/95.0%,) incrementing to 450…
Skipped 50 duplicate generated test vectors.
Minimum coverage not met (4.0%/95.0%,) incrementing to 500…
Skipped 50 duplicate generated test vectors.
Minimum coverage not met (4.0%/95.0%,) incrementing to 550…
Skipped 50 duplicate generated test vectors.
Minimum coverage not met (4.0%/95.0%,) incrementing to 600…
Skipped 50 duplicate generated test vectors.
Minimum coverage not met (4.0%/95.0%,) incrementing to 650…
Skipped 50 duplicate generated test vectors.
Minimum coverage not met (4.0%/95.0%,) incrementing to 700…
Skipped 50 duplicate generated test vectors.
Minimum coverage not met (4.0%/95.0%,) incrementing to 750…
Skipped 50 duplicate generated test vectors.
Minimum coverage not met (4.0%/95.0%,) incrementing to 800…
Skipped 50 duplicate generated test vectors.
Minimum coverage not met (4.0%/95.0%,) incrementing to 850…
Skipped 50 duplicate generated test vectors.
Minimum coverage not met (4.0%/95.0%,) incrementing to 900…
Skipped 50 duplicate generated test vectors.
Minimum coverage not met (4.0%/95.0%,) incrementing to 950…
Skipped 50 duplicate generated test vectors.
Minimum coverage not met (4.0%/95.0%,) incrementing to 1000…
Skipped 50 duplicate generated test vectors.
Hit ceiling. Settling for current coverage.
Time elapsed: 2.67s.
Simulations concluded: Coverage 4.0%

```
One more thing I would like to mention that I am using sky130_fd_sc_hd__tt_100C_1v80.lib as a lib file.

Manarabdelaty, 2021-01-04:
Can you post spm.netlist.v.cut.v ?

mahmoodulhassan-lm, 2021-01-04:
> Can you post spm.netlist.v.cut.v ?
Please find both .netlist and netlist.cut.

[spm.netlist.v.cut.txt](https://github.com/Cloud-V/Fault/files/5764776/spm.netlist.v.cut.txt)
[spm.netlist.txt](https://github.com/Cloud-V/Fault/files/5764775/spm.netlist.txt)

mahmoodulhassan-lm, 2021-01-06:
There is an issue with chain insertion commands as well.
```
fault chain -l Tech/sky130A/libs.ref/sky130_fd_sc_hd/lib/sky130_fd_sc_hd__tt_025C_1v80.lib -c Tech/sky130A/libs.ref/sky130_fd_sc_hd/verilog/sky130_fd_sc_hd.v --clock clk --reset rst Netlists/spm.netlist.v
Generating LALR tables
WARNING: 183 shift/reduce conflicts
Chaining internal flip-flops…
Internal scan chain successfuly constructed. Length:  0
Creating and chaining boundary flip-flops…
Generating LALR tables
WARNING: 183 shift/reduce conflicts
Boundary scan cells successfuly chained. Length:  3
Total scan-chain length:  3
Resynthesizing with yosys…
Verifying scan chain integrity…
Generating LALR tables
WARNING: 183 shift/reduce conflicts
done
Scan chain verification failed.
・Ensure that clock and reset signals, if they exist are passed as such to the program.
・Ensure that the reset is active high- pass --activeLow for activeLow.
・Ensure that D flip-flop cell name starts with ["DFFSR", "DFFPOSX1", "DFFNEGX1"].
・Ensure that there are no other asynchronous resets anywhere in the circuit.
Done.
```
In sky130_fd_sc_hd.v, cells definitions are different from Osu35. For example, flipflops names start with sky130_fd_sc_hd__udp_dff$PR_pp$PG$N etc. 
Please do correct me if I am understanding it wrong.

Manarabdelaty, 2021-01-11:
@mahmoodulhassan-lm

- I noticed that the synthesized netlist only has the CSADD module. To generate synthesized netlist with the SPM as top module, you need to specify the top module name using the -t option as follows 

  ` fault synth -l /ef/tech/SW/sky130A/libs.ref/sky130_fd_sc_hd/lib/sky130_fd_sc_hd__ff_100C_1v65.lib  -t SPM -o spm.synthesis.v spm.v`
  (You will need to replace the liberty file path with your own path)

- The low coverage you are exhibiting in the simulation is because the cut netlist wasn't generated correctly. The cut option should remove all flip-flop cells from the design. If you examined the netlist, you will see that the netlist still has flip-flop cells (I think we need to add a warning message for failing to detect any flip-flop cells)  To specify the flip-flop cell name, you can use -d option as follows: 

   `fault cut -d sky130_fd_sc_hd__dfrtp_1 spm.synthesis.v`
  (You will need to make sure that the synthesized netlist indeed has flip-flop cells named ` sky130_fd_sc_hd__dfrtp_1`; also note that this name will vary from design to design) 
  
I have attached the synthesized and the cut netlists I generated using above commands for your reference. 
[spm.synthesis.txt](https://github.com/Cloud-V/Fault/files/5795599/spm.synthesis.txt)
[spm.synthesis.v.cut.txt](https://github.com/Cloud-V/Fault/files/5795600/spm.synthesis.v.cut.txt)

closed 2021-02-08 by mahmoodulhassan-lm