AUCOHL/Fault #14: Fault switches not working

Prince-Joy, 2021-02-11:
@donn 
I am facing issues with fault, after completion of the installation procedure, when I am trying to get version using fault --version/-V
It's showing me the help menu every time, not only that,  only subcommand are working like
1. fault synth --top xyz -l xyz.lib xyz.v > its working
2. fault cut xyz.netlist.v > its also working but when I am trying to run **fault --cellModel xyz.v filename** its not working instead showing help menu every time.
[image]
[image]

donn, 2021-02-11:
If you check the resulting help menu, you will notice that one other option is required, namely --clock. 

Please add the missing option and retry.

Prince-Joy, 2021-02-12:
Thank you very much @donn 
facing another problem when running can you please look into it.
[image]
after that I have tried to install **iverilog** using the given make file **iverilog_build.swift** but showing the following errors
[image]

Prince-Joy, 2021-02-12:
Thank you so much @donn , I am facing another issue can you please look into it.
[image]

donn, 2021-02-13:
For the first issue, try updating IcarusVerilog, I recall having this issue myself with older versions.

Tap is @Manarabdelaty territory. Manar?

Manarabdelaty, 2021-02-13:
@Prince-Joy Can you make sure of the following ? 
- if the reset signal is asynchronous, you will need to add it to the ignored inputs during simulation; using `--ignoring`.This is because generating a test pattern for an asynchronous reset could arbitrarily reset the circuit if it is set to either 0 or  1. (You will need to specify the --ignoring for the main option and and both the chain and tap sub-commands)

If this is not the case with your design, can you post the chained.v file and .bin files ?

Prince-Joy, 2021-02-15:
@donn In my code reset is synchronous,
[image]
reference: asic-world
After running the following commands
[image]
I got the **error**:
[image]

PiyushSaini97, 2021-06-29:
@Prince-Joy I'm facing similar issues. When I try to run anything using osu035, I get the same at Test Vector generation. This is bypassed when nangate is used. However, at `fault asm` I encounter the mismatch of json and scan chain registers. I tried ignoring clock and reset with the help of `fault -i` command but the same error. Also, there seems to be a mismatch of just 1 input and scan register each time.

PiyushSaini97, 2021-06-30:
@Prince-Joy I found the reason for mismatch. Please ensure that you ignore your reset signal for TV generation step by appending `-i reset` option to `fault` command (your clock automatically gets ignored when `--clock` command is used. While running `fault chain` command, you are ignoring reset and clock using `--reset` and `--clock` respectively. If you had not ignored this at TV generation step then you basically got one extra test vector input as compared to the total scan chain register. 

@donn please verify whether this explanation is correct or not.

donn, 2021-06-30:
That's actually correct, yeah. You're supposed to add the reset to `-i`.

Maybe for consistency, I should add the `--reset` option to the `fault` command.

PiyushSaini97, 2021-06-30:
But wouldn't that cause issue for the designs that don't necessarily use reset?
For example, in the folder /Benchmarks/ISCAS_89, some of the designs don't use reset. I tried running s298.v. 
The module does not contain reset(or none of the inputs resemble a reset signal).

[image]

I tried creating TV for the same but ran into an error. And without much idea I cant consider any of the input as reset.
[image]

Please clarify if I misunderstood.

donn, 2021-06-30:
Good point, though the thing is I don't think it would be easy to make scan chains that cannot be reset. I'll have to discuss with the team whether we want to support no-reset designs.

donn, 2024-07-26:
So, I've decided to make most commands take `--clock` and `--reset`. There'll be an update in the future where we'll create the reset port if it doesn't exist.

closed 2024-07-26 by donn