AUCOHL/Fault #15: Fault is not accepting Synopysis Design Compiler generated netlist

mahmoodulhassan-lm, 2021-02-19:
I am trying to run FAULT on Swerve el-2 design. I have DC synthesized netlist which is mapped on Sky 130 A library. When I try to run the cut command following error appears.  
Command: 
`
fault cut -d sky130_fd_sc_hd__dfrtp_1  Netlists/netlist.v`

Output:
```
Generating LALR tables
WARNING: 183 shift/reduce conflicts
Fatal error: 'try!' expression unexpectedly raised an error: Python exception:  line:23: before: "."
Traceback:
  File "/usr/local/lib/python3.6/dist-packages/pyverilog/vparser/parser.py", line 2338, in parse
    ast = codeparser.parse()
  File "/usr/local/lib/python3.6/dist-packages/pyverilog/vparser/parser.py", line 2316, in parse
    ast = self.parser.parse(text, debug=debug)
  File "/usr/local/lib/python3.6/dist-packages/pyverilog/vparser/parser.py", line 77, in parse
    return self.parser.parse(text, lexer=self.lexer, debug=debug)
  File "/usr/local/lib/python3.6/dist-packages/ply/yacc.py", line 333, in parse
    return self.parseopt_notrack(input, lexer, debug, tracking, tokenfunc)
  File "/usr/local/lib/python3.6/dist-packages/ply/yacc.py", line 1201, in parseopt_notrack
    tok = call_errorfunc(self.errorfunc, errtoken, self)
  File "/usr/local/lib/python3.6/dist-packages/ply/yacc.py", line 192, in call_errorfunc
    r = errorfunc(token)
  File "/usr/local/lib/python3.6/dist-packages/pyverilog/vparser/parser.py", line 2268, in p_error
    self._raise_error(p)
  File "/usr/local/lib/python3.6/dist-packages/pyverilog/vparser/parser.py", line 2279, in _raise_error
    raise ParseError("%s: %s" % (coord, msg))
: file PythonKit/Python.swift, line 602
Current stack trace:
0    libswiftCore.so                    0x00007fb46ba62990 swift_reportError + 50
1    libswiftCore.so                    0x00007fb46bad6260 _swift_stdlib_reportFatalErrorInFile + 115
2    libswiftCore.so                    0x00007fb46b7bb925 <unavailable> + 1399077
3    libswiftCore.so                    0x00007fb46b7bb567 <unavailable> + 1398119
4    libswiftCore.so                    0x00007fb46b7bbb02 <unavailable> + 1399554
5    libswiftCore.so                    0x00007fb46b7b9fa0 _assertionFailure(_:_:file:line:flags:) + 517
6    libswiftCore.so                    0x00007fb46b807859 <unavailable> + 1710169
7    fault                              0x000056190d8e926a <unavailable> + 1466986
8    fault                              0x000056190d859f06 <unavailable> + 880390
9    fault                              0x000056190d876413 <unavailable> + 996371
10   fault                              0x000056190d876511 <unavailable> + 996625
11   fault                              0x000056190d883853 <unavailable> + 1050707
12   fault                              0x000056190d8762be <unavailable> + 996030
13   libc.so.6                          0x00007fb469c96b10 __libc_start_main + 231
14   fault                              0x000056190d7a37fa <unavailable> + 133114
Illegal instruction (core dumped)
[netlist.txt](https://github.com/Cloud-V/Fault/files/6010072/netlist.txt)

```
I have attached the netlist herewith.

Manarabdelaty, 2021-02-19:
The issue seems to be that [pyverilog](https://github.com/PyHDI/Pyverilog) can't parse the `.ic_wr_data({....})` in the module header.

Manarabdelaty, 2021-02-19:
@mahmoodulhassan-lm I filed an issue for that at https://github.com/PyHDI/Pyverilog/issues/80

donn, 2021-02-20:
> @mahmoodulhassan-lm I filed an issue for that at [PyHDI/Pyverilog#80](https://github.com/PyHDI/Pyverilog/issues/80)

Pyverilog has a lot of Verilog missing unfortunately. Doesn't help that its a Python Lex Yacc project which means I hate myself trying to contribute to it.

I'll take a look but no promises.

closed 2021-04-15 by mahmoodulhassan-lm