AUCOHL/Fault #16: Installation error

PiyushSaini97, 2021-06-26:
Hi,
I'm trying to install Fault on Windows Subsystem for Linux (WSL2). I installed prerequisites for the software and all of them are working fine. However, when I try to install Fault in my system, it gives me an error. Find the image below-

[image]
 
I don't understand what the error is about.

Note: I did not install Atalanta or PODEM as they were optional and thought to use the software's pattern generation 

Please help.

Regards,
Piyush

Zhaowenqing, 2021-06-26:
> Hi,
> I'm trying to install Fault on Windows Subsystem for Linux (WSL2). I installed prerequisites for the software and all of them are working fine. However, when I try to install Fault in my system, [...]

I  also encountered this problem. Then I changed the swift version to 5.2, and the problem was solved.You can try this method

donn, 2021-06-26:
@Zhaowenqing is correct: Fault supports Swift 5.2+ as shown in [INSTALLING.md](https://github.com/Cloud-V/Fault/blob/d59d45d307f48a5b3082571507a589ccc9beafbc/INSTALLING.md).

I'll add like a little GitHub badge to emphasize this in the Readme.

donn, 2021-06-26:
Done.

PiyushSaini97, 2021-06-27:
I was using the latest version of swift(5.4.1) for installation purpose and encountered this error. Is it that the software can be built using only swift 5.2?

donn, 2021-06-27:
Um, not really. That's quite interesting. I'll have to take another look.

PiyushSaini97, 2021-06-28:
Hi. I was able to install the software this time with the help of swift 5.3. I had to move from WSL to ubuntu OS for the installation.
While installing I ran into two other problems. 
1) I was not able to install the atalanta and podem files. It gave me error of URLexecutable.
2) After going through all the steps given in https://github.com/Cloud-V/Fault/blob/d59d45d307f48a5b3082571507a589ccc9beafbc/INSTALLING.md, I was shown a message "installed". But when I tried running the software, my system prompted "fault not found". I double checked the bin folder to ensure that it was installed.

[image]

Please help.

donn, 2021-06-28:
The warnings are expected.

The issue is actually what you see right after Swift's warnings: `sh: 1: curl: not found`. You need to install curl: `sudo apt-get install curl`. There's a slight API deviation between Linux and macOS in Swift that requires Linux to use a slightly older API, triggering those warnings.

As for Fault not being found, that's because your `~/bin` folder is not in PATH. You'll need to add `export PATH=$HOME/bin:$PATH;` to your `~/.profile`. 

Alternatively, you can do `INSTALL_DIR=/usr/local/bin swift install.swift` to install to a folder known to be in PATH.

As an aside, you're invoking the Atalanta/PODEM installers using EXEC_PREFIX=</usr/local/bin>. This is not a valid path: the tutorial shows the `<>` to indicate that this can be changed.

PiyushSaini97, 2021-06-28:
Thank you for your time. Fault installed perfectly. But this time I encountered another problem. When I tried `fault --help`, it prompted me to invoke it again. 

[image]

I encountered this for `fault synth --help` and `fault tap --help` as well. Other help commands worked.

Also, when I tried installing atalanta once again it gave me another error (referenced in the image).

donn, 2021-06-30:
Confirmed issue. I think I botched it while trying to enforce "required" flags...

PiyushSaini97, 2021-06-30:
So should I reinstall Fault in order to use the updated one?

donn, 2021-06-30:
Give me a moment to push the update.

closed 2021-06-30 by donn

PiyushSaini97, 2021-06-30:
Ok. Thank you for your time.

commit 1d1233be (2021-06-30): Confirm Swift 5.4 Compatibility, Fix #16