AUCOHL/Fault #19: Scan chain Verification failed

PiyushSaini97, 2021-06-30:
Hi,
While running `fault chain` command I bumped into an error. Please refer image.

[image]

I generated TV with default options. The prompt displays "Scan chain verification failed". I'm confused, which program is being referenced here that requires clock and reset? The netlist generated using yosys contained reset and clock signals.
The reset in my design is synchronous(and is the only reset signal).
Even with all these errors, **.chained** file is created in my netlist folder. I'm able to run `tap asm` command and check the created **.bin** files. 
I encountered similar issue while running `fault tap` command. **.jtag** file was created but the prompt displayed "Tap port verification failed".

[image]

Also, I'm curious to know what would have been the output if both these steps verified successfully. 

Regards,
Piyush

donn, 2021-07-01:
That scan chain verification just verifies if the created scan chain works properly. That means it failed. The tap also fails because scan chain verification fails. It doesn't affect the creation of the scan chain itself.

I'll have to test and get back to you on why counter is failing scan chain verification. Thank you for your patience and diligence in reporting bugs. :)

PiyushSaini97, 2021-07-01:
Not only counter. I tried running universal 8 bit shift register.

Code:
[image]

Unfortunately I encountered the same error :(

PiyushSaini97, 2021-07-02:
@donn I understood the problem. I was using asynchronous reset in both of my designs. For that reason my scan chain verification failed. After changing that my scan chain verified successfully.

However, now I bumped into another error. I ran `fault tap` and got this 
[image]

But now if I append `--skipSynth` to it then I get this
[image]

Tap port verified successfully in this but test vector simulation failed. 

According to the error I passed `--activeLow` but then I get this error

[image]

What is the reason I'm encountering these errors and why adding `--skipSynth` verified the tap port successully?

bharath19-gs, 2023-01-23:
@donn 
any solution for this issue?
i am facing the same issue too.

please do give your valuable inputs