AUCOHL/Fault #20: Swift not found

IshaShah2806, 2021-07-01:
I was installing Fault and I had already installed Swift in my Ubuntu 20.04 yet while installing Fault I faced an error SWIFT not found.
[image]

donn, 2021-07-01:
Are you using WSL or WSL2?

At any rate it's curious that `sh` drops PATH. Can you replace `let compilationResult = "swift build".shOutput()` with `let compilationResult = "echo $PATH".shOutput() && false` and let me know the output?

IshaShah2806, 2021-07-01:
Thank you for your time .I am using WSL 2  and I made the necessary changes and I got this output.
[image]

donn, 2021-07-01:
Ah. Gotcha, always need to confirm no one is using WSL1. It's too unpredictable. :)

The issue is your PATH has a literal tilde, not your home folder. Can you replace `~/swift-5.3.3-RELEASE-ubuntu20.04/usr/bin` with an absolute path where you're setting the PATH variable? In this case, it would be `/home/isha/swift-5.3.3-RELEASE-ubuntu20.04/usr/bin`.

IshaShah2806, 2021-07-01:
I replaced as directed .But still the same error
[image]
[image]

donn, 2021-07-01:
That's so weird. I'll have to come back to this later- we'll probably need a workaround.

RameenAnwar, 2022-04-12:
try `export PATH=home/isha/Fault/swift-5.3.3-RELEASE-ubuntu20.04/usr/bin`
then `swift ./install.swift`

donn, 2023-01-13:
For future searchers: please use the AppImage.

closed 2023-01-13 by donn