AUCOHL/Fault #22: Query - Simulating Pure combinational design

PiyushSaini97, 2021-07-01:
Hi,

I was running the design c17.v in `Benchmarks/ISCAS_85`. However, while generating TV, the command requires me to add clock as well and obviously a combinational circuit does not have clock.

Verilog Code:

[image]

My question: Is it possible to generate TV adnd find faults for pure combinational designs?

priyajit99, 2022-06-05:
Same issue .

donn, 2022-06-08:
I think you can just pass `--clock NULL`. I can't recall why we require`clock` at the moment, but I do recall it not being required in an earlier version.

priyajit99, 2022-06-08:
It is showing this error (attached).

donn, 2022-06-08:
I'm not seeing any attachment. Can you please reply on GitHub?

priyajit99, 2022-06-08:
It is showing this error
[image]

donn, 2022-06-08:
Ugh... yeah, that tracks. Okay. I'll look into it.

donn, 2022-06-08:
You may want to consider adding a dummy clock port that does nothing for now.

donn, 2022-06-08:
Alright, so I've put in the research.

Fact is- combinational designs are a tiny minority of practical circuits. So, I've added instructions to the Wiki that just recommend adding a dummy clock ***and*** reset port to the design, which will be later used for the scan chain and JTAG TAP.

Sorry for the inconvenience. The alternative would again greatly complicate the codebase, at the cost of adding two unused inputs. It's not worth it.

closed 2022-06-08 by donn