AUCOHL/Fault #30: Missing external test set example file s27.v.tv.json

padiyarsandesh, 2022-06-27:
Hi,
The external test vector example file s27.v.tv.json ( PATH :RTL/ISCAS_89/s27.v.tv.json) is missing. 
Would appreciate if the file would be available as I want to use external test set for my research purposes.

Thanks

donn, 2022-06-27:
Do we link to that anywhere?

padiyarsandesh, 2022-06-27:
Yeah , its there is the wiki usage , under section: 2)Input TV Set a)JSON Format
[image]

priyajit99, 2023-01-12:
Same issue here . @donn pls look into the issue for example test pattern json file.

cmarqu, 2023-01-12:
The file apparently got lost in https://github.com/AUCOHL/Fault/commit/cd066bc1478ab8cf6795c2df3442316b5e7960d5 but can be extracted from the history: https://github.com/AUCOHL/Fault/blob/432adf445acef670fed994950fcf92c8cf8cfefa/RTL/ISCAS_89/s27.v.tv.json

priyajit99, 2023-01-12:
On the file it is showing invalid file

donn, 2023-01-13:
Well, for various reasons, the vector format may or may not have changed. I would assume this is an issue of outdated documentation and not an issue of the file being invalid.

I will update the documentation.

closed 2023-01-13 by donn