AUCOHL/Fault #39: Error message "Option --clock is required." when run Fault Simulation

TonyHo722, 2023-07-24:
I reference below link 
https://github.com/AUCOHL/Fault/wiki/Usage

in Fault Simulations, I found error log as below
```
(base) tonyho@ubuntu5:~/workspace/fault/Fault$ fault -c Tech/osu035/osu035_stdcells.v -v 100 -r 50 -m 95 --ceiling 1000 Netlists/s27.netlist.v.cut.v 
Option --clock is required.
Invoke fault --help for more info.
```
after try it, I add "--clock CK" then pass as below. 
I think document need update.
```
(base) tonyho@ubuntu5:~/workspace/fault/Fault$ fault -c Tech/osu035/osu035_stdcells.v -v 100 -r 50 -m 95 --ceiling 1000 Netlists/s27.netlist.v.cut.v --clock CK
Generating LALR tables
WARNING: 183 shift/reduce conflicts
Processing module s27…
Found 48 fault sites in 12 gates and 15 ports.
Performing simulations…
Skipped 4 duplicate generated test vectors.
Minimum coverage not met (81.25%/95.0%,) incrementing to 150…
Skipped 4 duplicate generated test vectors.
Minimum coverage not met (81.25%/95.0%,) incrementing to 200…
Skipped 11 duplicate generated test vectors.
Minimum coverage not met (81.25%/95.0%,) incrementing to 250…
Skipped 10 duplicate generated test vectors.
Minimum coverage not met (81.25%/95.0%,) incrementing to 300…
Skipped 11 duplicate generated test vectors.
Minimum coverage not met (81.25%/95.0%,) incrementing to 350…
Skipped 9 duplicate generated test vectors.
Minimum coverage not met (81.25%/95.0%,) incrementing to 400…
Skipped 17 duplicate generated test vectors.
Minimum coverage not met (81.25%/95.0%,) incrementing to 450…
Skipped 14 duplicate generated test vectors.
Minimum coverage not met (81.25%/95.0%,) incrementing to 500…
Skipped 15 duplicate generated test vectors.
Minimum coverage not met (81.25%/95.0%,) incrementing to 550…
Skipped 17 duplicate generated test vectors.
Minimum coverage not met (81.25%/95.0%,) incrementing to 600…
Skipped 16 duplicate generated test vectors.
Minimum coverage not met (81.25%/95.0%,) incrementing to 650…
Skipped 26 duplicate generated test vectors.
Minimum coverage not met (81.25%/95.0%,) incrementing to 700…
Skipped 30 duplicate generated test vectors.
Minimum coverage not met (81.25%/95.0%,) incrementing to 750…
Skipped 27 duplicate generated test vectors.
Minimum coverage not met (81.25%/95.0%,) incrementing to 800…
Skipped 28 duplicate generated test vectors.
Minimum coverage not met (81.25%/95.0%,) incrementing to 850…
Skipped 28 duplicate generated test vectors.
Minimum coverage not met (81.25%/95.0%,) incrementing to 900…
Skipped 26 duplicate generated test vectors.
Minimum coverage not met (81.25%/95.0%,) incrementing to 950…
Skipped 27 duplicate generated test vectors.
Minimum coverage not met (81.25%/95.0%,) incrementing to 1000…
Skipped 38 duplicate generated test vectors.
Hit ceiling. Settling for current coverage.
Time elapsed: 6.14s.
Simulations concluded: Coverage 81.25%
```

TonyHo722, 2023-07-24:
Note: the CK is come from s27 module input signal for clock is CK in Netlists/s27.netlist.v.cut.v.

module s27
(
  GND,
  VDD,
  CK,
  reset,
  G0,
  G1,
  G17,
  G2,
  G3,
  _20_,
  \_20_.q ,
  _21_,
  \_21_.q ,
  _22_,
  \_22_.q
);

rafidmuttaki, 2024-05-24:
[image]
I am getting this issue for the above command. Not sure what might be wrong?

rafidmuttaki, 2024-05-29:
> ![image](https://private-user-images.githubusercontent.com/63279820/333629955-5ee5054f-1607-45e6-8d4b-faf122d30035.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

can someone please help with this?

donn, 2024-07-26:
Please try the new documentation. Sorry for the inconvenience.

closed 2024-07-26 by donn